TheSpaghettiDetective/obico-server #995: WEB Interface Loading Issue with Docker-Compose Setup: Uncaught TypeError in page-context.js

7010G, 2024-07-24:
# WEB Interface Loading Issue with Docker-Compose Setup: Uncaught TypeError in page-context.js

## Docker Environment Information:
```shell
root@nas7010:/volume1/Download/obico-server# docker -v
Docker version 20.10.3, build 55f0773
root@nas7010:/volume1/Download/obico-server# docker-compose -v
Docker Compose version v2.29.0
root@nas7010:/volume1/Download/obico-server#
```
## Build Log:
```shell
root@nas7010:/volume1/Download/obico-server# docker-compose up --build -d
WARN[0000] /volume1/Download/obico-server/docker-compose.yml: the attribute `version` is obsolete, it will be ignored, please remove it to avoid potential confusion
[+] Building 24.8s (28/28) FINISHED                                                                                                              docker:default
 => [tasks internal] load build definition from Dockerfile                                                                                                 2.4s
 => => transferring dockerfile: 32B                                                                                                                        0.0s
 => [tasks internal] load .dockerignore                                                                                                                    3.2s
 => => transferring context: 34B                                                                                                                           0.0s
 => [ml_api internal] load build definition from Dockerfile                                                                                                4.2s
 => => transferring dockerfile: 32B                                                                                                                        0.0s
 => [ml_api internal] load .dockerignore                                                                                                                   4.8s
 => => transferring context: 34B                                                                                                                           0.0s
 => [web internal] load metadata for docker.io/thespaghettidetective/web:base-1.16                                                                         0.0s
 => [tasks internal] load build context                                                                                                                    1.9s
 => => transferring context: 32.22kB                                                                                                                       0.0s
 => [web 1/5] FROM docker.io/thespaghettidetective/web:base-1.16                                                                                           0.0s
 => [ml_api internal] load metadata for docker.io/thespaghettidetective/ml_api_base:1.4                                                                    0.0s
 => [ml_api 1/9] FROM docker.io/thespaghettidetective/ml_api_base:1.4                                                                                      0.0s
 => [ml_api internal] load build context                                                                                                                   2.0s
 => => transferring context: 879B                                                                                                                          0.0s
 => CACHED [web 2/5] WORKDIR /app                                                                                                                          0.0s
 => CACHED [web 3/5] RUN pip install -U pip pipenv==2022.12.19                                                                                             0.0s
 => CACHED [tasks 4/5] ADD ./ /app                                                                                                                         0.0s
 => CACHED [tasks 5/5] RUN pip install -r requirements.txt                                                                                                 0.2s
 => [web] exporting to image                                                                                                                              11.6s
 => => exporting layers                                                                                                                                    0.0s
 => => writing image sha256:84645eedb9687438407bbdf6bc0feb58c4ea4513c9d8eaf98915dd2025dc7c46                                                               0.1s
 => => naming to docker.io/library/obico-server-tasks                                                                                                      0.0s
 => => writing image sha256:6513da29c20111e185d1c75dec5995992547790892233c300e535a2849cf53ab                                                               0.0s
 => => naming to docker.io/library/obico-server-ml_api                                                                                                     0.0s
 => => writing image sha256:4d8b1389271a43fe15a98101a8a2f3ddf8d162504d31ee4cf8b4cd45a08ac419                                                               0.0s
 => => naming to docker.io/library/obico-server-web                                                                                                        0.0s
 => CACHED [ml_api 2/9] WORKDIR /app                                                                                                                       0.0s
 => CACHED [ml_api 3/9] ADD . /app                                                                                                                         0.0s
 => CACHED [ml_api 4/9] RUN pip install --upgrade pip                                                                                                      0.0s
 => CACHED [ml_api 5/9] RUN pip install -r requirements.txt                                                                                                0.0s
 => CACHED [ml_api 6/9] RUN echo 'Downloading the latest failure detection AI model in Darknet format...'                                                  0.0s
 => CACHED [ml_api 7/9] RUN wget -O model/model-weights.darknet $(cat model/model-weights.darknet.url | tr -d '\r')                                        0.0s
 => CACHED [ml_api 8/9] RUN echo 'Downloading the latest failure detection AI model in ONNX format...'                                                     0.0s
 => CACHED [ml_api 9/9] RUN wget -O model/model-weights.onnx $(cat model/model-weights.onnx.url | tr -d '\r')                                              0.4s
 => [web internal] load build definition from Dockerfile                                                                                                   2.4s
 => => transferring dockerfile: 32B                                                                                                                        0.0s
 => [web internal] load .dockerignore                                                                                                                      3.2s
 => => transferring context: 34B                                                                                                                           0.0s
 => [web internal] load build context                                                                                                                      1.0s
 => => transferring context: 32.22kB                                                                                                                       0.0s
 => CACHED [web 4/5] ADD ./ /app                                                                                                                           0.0s
 => CACHED [web 5/5] RUN pip install -r requirements.txt                                                                                                   0.0s
[+] Running 4/4
 ✔ Container obico-server-ml_api-1  Running                                                                                                                0.0s
 ✔ Container obico-server-redis-1   Running                                                                                                                0.0s
 ✔ Container obico-server-web-1     Started                                                                                                               14.7s
 ✔ Container obico-server-tasks-1   Started 
```
## Issue Description:
[... cut ...]
    at l (bootstrap:89:22)
    at i (bootstrap:45:15)
    at Array.t [as push] (bootstrap:32:11)
    at chunk-vendors.28dd249032e4.js:1:53
```
## IThe problematic code is:
```shell
export const mobilePlatform = () => {
  return JSON.parse(document.querySelector('#app-platform-json').text)['platform']
}
```
I can access http://100.115.23.128:3334/admin/ normally, but http://100.115.23.128:3334/printers/ fails to load.
[image]
[image]

I'm puzzled and don't know where the problem is, as the backend logs appear normal.

kennethjiang, 2024-07-24:
It looks like somehow your frontend and backend is out of sync, as the code you saw as outdated by 4c4ef680e7b92beba3407a81712fce6b1a73ae88

7010G, 2024-07-24:
> It looks like somehow your frontend and backend is out of sync, as the code you saw as outdated by [4c4ef68](https://github.com/TheSpaghettiDetective/obico-server/commit/4c4ef680e7b92beba3407a81712fce6b1a73ae88)

Thanks Maybe it's a problem with my mirror repository

kennethjiang, 2024-07-24:
Ok. Please update your server. If the problem is fixed after the update, please close this issue.

7010G, 2024-07-24:
> Ok. Please update your server. If the problem is fixed after the update, please close this issue.

After using https://github.com/gabe565/docker-obico.git to build image of my problem is resolved

closed 2024-07-24 by 7010G