TheSpaghettiDetective/obico-server #1011: [BUG] Upgrade from 0b2da44 to current results in Server Error (500)

querex007, 2024-08-10:
**Note**
Most of the times, an issue you may potentially run into is not a bug. Instead, it is related to configurations, such as firewall rules, incompatible docker version, reverse proxy setups, etc.

If you are not sure, please don't open a bug report here. Instead, you can get the issue solved much faster by seeking help in our support forum: https://obico.io/discord.

Proceed to opening a bug report only if you are sure it is a defect in the code given a correct configuration.

**Describe the bug**
Upgrade from 0b2da44 to current results in Server Error (500)

**To Reproduce**
Steps to reproduce the behavior:
1. Set up ennvironment with release 0b2da44
2. Perform upgrade as documentation says: git checkout release && git pull && docker compose up --build -d
3. Opening web GUI on port 3334 results into a "Server Error (500)"

**Log files in attachment**
[_obico-server-ml_api-1_logs.txt](https://github.com/user-attachments/files/16568878/_obico-server-ml_api-1_logs.txt)
[_obico-server-web-1_logs.txt](https://github.com/user-attachments/files/16568879/_obico-server-web-1_logs.txt)

appleimperio, 2024-08-12:
Try this

````
cd obico-server && docker compose run web ./manage.py site --add ip of the Proxmox VM
````

There is more info on the Django server configuration https://www.obico.io/docs/server-guides/configure/

querex007, 2024-08-13:
@appleimperio that fixed the issue!
Many thanks.
Will close as resolved.

closed 2024-08-13 by querex007