TheSpaghettiDetective/obico-server #1018: Server Error (500)

Dudgemo, 2024-08-24:
Trying to install on unraid docker, following exact instructions on Obico how to and I get a Server Error (500) when navigating to the web server. 

I'm not experienced with docker compose and I'm stuck on what to do next.

See below log.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/asgiref/sync.py", line 534, in thread_handler
    raise exc_info[1]
  File "/usr/local/lib/python3.10/site-packages/django/core/handlers/exception.py", line 43, in inner
    response = await get_response(request)
  File "/usr/local/lib/python3.10/site-packages/django/core/handlers/base.py", line 253, in _get_response_async
    response = await wrapped_callback(
  File "/usr/local/lib/python3.10/site-packages/asgiref/sync.py", line 479, in __call__
    ret: _R = await loop.run_in_executor(
  File "/usr/local/lib/python3.10/site-packages/asgiref/current_thread_executor.py", line 40, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/local/lib/python3.10/site-packages/asgiref/sync.py", line 538, in thread_handler
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/django/views/generic/base.py", line 84, in view
    return self.dispatch(request, *args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/django/utils/decorators.py", line 46, in _wrapper
    return bound_method(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/django/views/decorators/debug.py", line 92, in sensitive_post_parameters_wrapper
    return view(request, *args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/django/utils/decorators.py", line 46, in _wrapper
    return bound_method(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/django/views/decorators/cache.py", line 62, in _wrapped_view_func
    response = view_func(request, *args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/allauth/account/views.py", line 84, in dispatch
    return super().dispatch(request, *args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/allauth/account/mixins.py", line 40, in dispatch
    response = super().dispatch(request, *args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/django/views/generic/base.py", line 119, in dispatch
    return handler(request, *args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/allauth/account/mixins.py", line 59, in get
    response = super().get(request, *args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/django/views/generic/edit.py", line 144, in get
    return self.render_to_response(self.get_context_data())
  File "/usr/local/lib/python3.10/site-packages/allauth/account/views.py", line 106, in get_context_data
    site = get_current_site(self.request)
  File "/usr/local/lib/python3.10/site-packages/django/contrib/sites/shortcuts.py", line 14, in get_current_site
    return Site.objects.get_current(request)
  File "/usr/local/lib/python3.10/site-packages/django/contrib/sites/models.py", line 61, in get_current
    return self._get_site_by_request(request)
  File "/usr/local/lib/python3.10/site-packages/django/contrib/sites/models.py", line 45, in _get_site_by_request
    SITE_CACHE[domain] = self.get(domain__iexact=domain)
  File "/usr/local/lib/python3.10/site-packages/django/db/models/manager.py", line 85, in manager_method
    return getattr(self.get_queryset(), name)(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/django/db/models/query.py", line 496, in get
    raise self.model.DoesNotExist(
django.contrib.sites.models.Site.DoesNotExist: Site matching query does not exist.

Dudgemo, 2024-08-24:
Cheers, that worked straight away. Obviously this step needs to be added for the unraid install.

kennethjiang, 2024-08-25:
@WhiskeySheppard thank you for pointing out that the help doc may be misleading. Fixed it with https://github.com/TheSpaghettiDetective/obico-server/commit/d2ba7b9b51fc0f7df2ee8a6361d3d185a560e40e

closed 2024-08-25 by kennethjiang