TheSpaghettiDetective/obico-server #1023: No video stream, but detection still works.

purplepotables, 2024-09-11:
**Note**
Most of the times, an issue you may potentially run into is not a bug. Instead, it is related to configurations, such as firewall rules, incompatible docker version, reverse proxy setups, etc.

If you are not sure, please don't open a bug report here. Instead, you can get the issue solved much faster by seeking help in our support forum: https://obico.io/discord.

Proceed to opening a bug report only if you are sure it is a defect in the code given a correct configuration.

**Describe the bug**
I am running obico in a docker container in an lxc in proxmox. I have also added my docker-compose yaml to it to spin up octoprint and obico at the same time. I have tried running octoprint from a raspberry pi, a debian VM, and a separate docker host. The problem is the same in all of them. The setup acts like it works, but in obico I never get a video feed but everything else functions, and octo print will have a warning that it could not connect to the obico server. When I hit test it gives me a green check mark, but the error state stays. If I reboot my octo print the percent of connection failures goes down, but it still behaves the same. It also looks like print failure is detecting because I got screenshots of the printer in my notifications (between failure to stream notifications). The stream does work in octoprint in all tests.

**To Reproduce**
Steps to reproduce the behavior:
Run the environment provided and connect the systems.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hosting environment (please complete the following information):**
Debian container in proxmox hosting docker with an Nvidia GPU, the printer, and the camera passed through. Here is the docker compose:
root@hermes:/home/obico/obico-server# cat docker-compose.yml 
version: '2.4'

x-web-defaults: &web-defaults
  restart: unless-stopped
  build:
    context: backend
    dockerfile: 'Dockerfile'
  volumes:
    - ./backend:/app
    - ./frontend:/frontend
  depends_on:
    - redis
  environment:
    OCTOPRINT_TUNNEL_PORT_RANGE: '0-0'
    EMAIL_HOST: '${EMAIL_HOST-}'
    EMAIL_HOST_USER: '${EMAIL_HOST_USER-}'
    EMAIL_HOST_PASSWORD: '${EMAIL_HOST_PASSWORD-}'
    EMAIL_PORT: '${EMAIL_PORT-587}'
    EMAIL_USE_TLS: '${EMAIL_USE_TLS-True}'
    DEFAULT_FROM_EMAIL: '${DEFAULT_FROM_EMAIL-changeme@example.com}'
    DEBUG: '${DEBUG-False}'    # Don't set DEBUG to True unless you know what you are doing. Otherwise the static files will be cached in browser until hard-refresh
    ADMIN_IP_WHITELIST: '${ADMIN_IP_WHITELIST-}'
    SITE_USES_HTTPS: '${SITE_USES_HTTPS-False}'
    SITE_IS_PUBLIC: '${SITE_IS_PUBLIC-False}'
    CSRF_TRUSTED_ORIGINS: '${CSRF_TRUSTED_ORIGINS-}'
    SOCIAL_LOGIN: '${SOCIAL_LOGIN-False}'
    REDIS_URL: '${REDIS_URL-redis://redis:6379}'
    DATABASE_URL: '${DATABASE_URL-sqlite:////app/db.sqlite3}'
    INTERNAL_MEDIA_HOST: '${INTERNAL_MEDIA_HOST-http://web:3334}'
    ML_API_HOST: '${ML_API_HOST-http://ml_api:3333}'
    ACCOUNT_ALLOW_SIGN_UP: '${ACCOUNT_ALLOW_SIGN_UP-False}'
    WEBPACK_LOADER_ENABLED: '${WEBPACK_LOADER_ENABLED-False}'
    TELEGRAM_BOT_TOKEN: '${TELEGRAM_BOT_TOKEN-}'
    TWILIO_ACCOUNT_SID: '${TWILIO_ACCOUNT_SID-}'
    TWILIO_AUTH_TOKEN: '${TWILIO_AUTH_TOKEN-}'
    TWILIO_FROM_NUMBER: '${TWILIO_FROM_NUMBER-}'
    SENTRY_DSN: '${SENTRY_DSN-}'
    PUSHOVER_APP_TOKEN: '${PUSHOVER_APP_TOKEN-}'
    SLACK_CLIENT_ID: '${SLACK_CLIENT_ID-}'
    SLACK_CLIENT_SECRET: '${SLACK_CLIENT_SECRET-}'
    DJANGO_SECRET_KEY: '${DJANGO_SECRET_KEY-}'
    SYNDICATE: '${SYNDICATE-}'
    VERSION:
services:
  ml_api:
    hostname: ml_api
    restart: unless-stopped
    build:
      context: ml_api
    environment:
      DEBUG: 'True'
      FLASK_APP: 'server.py'
      # ML_API_TOKEN:
    tty: true
    command: bash -c "gunicorn --bind 0.0.0.0:3333 --workers 1 wsgi"

  web:
    <<: *web-defaults
    hostname: web
    ports:
      - "3334:3334"
    depends_on:
      - ml_api
    command: sh -c 'python manage.py migrate && python manage.py collectstatic -v 2 --noinput && daphne -b 0.0.0.0 -p 3334 config.routing:application'

  tasks:
    <<: *web-defaults
    hostname: tasks
    command: sh -c "celery -A config worker --beat -l info -c 2 -Q realtime,celery"

  redis:
    restart: unless-stopped
    image: redis:7.2-alpine

  octoprint:
    image: octoprint/octoprint
    restart: unless-stopped
    privileged: true
    ports:
      - 80:80
    devices:
      - /dev/ttyusb0:/dev/ttyusb0
      - /dev/bus/usb/001:/dev/bus/usb/001
      - /dev/video0:/dev/video0
      - /dev/video1:/dev/video1
#    devices:
#      - /dev/ttyUSB0:/dev/ttyUSB0
#      - /dev/video0:/dev/video0
##      - /dev/video1:/dev/video1
    volumes:
     - octoprint:/octoprint
     - /mnt/octo:/octoprint/octoFiles
#    # uncomment the lines below to ensure camera streaming is enabled when
#    # you add a video device
    environment:
      - ENABLE_MJPG_STREAMER=true
##      - MJPG_STREAMER_INPUT=-y -n -r 640x480 -vf
#  ####
  # uncomment if you wish to edit the configuration files of octoprint
  # refer to docs on configuration editing for more information
  ####

  #config-editor:
  #  image: linuxserver/code-server
  #  ports:
  #    - 8443:8443
  #  depends_on:
  #    - octoprint
  #  restart: unless-stopped
  #  environment:
  #    - PUID=0
  #    - PGID=0
  #    - TZ=America/Chicago

volumes:
  octoprint:

**Additional context**

I am connecting the two through the IP of the docker host.

OttersMeep, 2024-12-19:
I am having a similar issue and I think I may have found the problematic commit
47072e0e5b978e035680658acb4ef82f09c76e0f
[image]

ManuelW77, 2024-12-20:
Same here. Detection works, Obico works, but i can't see any images and videos in obico.
Inspecting the working directory shows that videos, timelapse etc is there.

kennethjiang, 2024-12-23:
@OttersMeep If you think you have found a bug, you can fix it and submit a PR. I'll review, test, and merge it. But my hunch is this is a server/network configuration issue.

OttersMeep, 2024-12-23:
@kennethjiang  Issue persists on fresh install, but works fine on a virtualized network. Will look through Linux/router settings.
Note: Most likely a router issue of some sort, newer AT&T routers don't like supporting protocols you don't explicitly tell them to

kennethjiang, 2024-12-23:
Ok I will close the ticket for now. We can re-open it if confirmed to be a bug.

Also our discord server is a good place to get help on diagnosing problems like this.

closed 2024-12-23 by kennethjiang