TheSpaghettiDetective/obico-server #1042: [BUG] Wrong pic.img_url if multiple Django site is set

whc2001, 2024-10-29:
**Describe the bug**
When multiple site is set up in the Django admin page, The printer's `pic.img_url` seems only use the first site entry, which can cause invalid thumbnails.

**To Reproduce**
Steps to reproduce the behavior:
1. Set up three sites in Django admin page: `obico.website.com:48877` (for main access behind reverse proxy, port 443 is blocked by ISP), `obico.website.com:15853` and `obico.website.com:15854` (for tunnel access)
2. Open `obico.website.com:48877`, everything works except printer webcam thumbnail
3. Check packet capture, the websocket backend give `pic.img_url` a value of `https://obico.website.com:15853/media/tsd-pics/snapshots/2/latest_unrotated.jpg?digest=...`, which should be `https://obico.website.com:48877/media...`

**Hosting environment (please complete the following information):**
 - OS: Debian
 - Version: 12
 - Hardware: KVM Virtual Machine

**Additional context**
After a brief looking I think it might be related to this:
https://github.com/TheSpaghettiDetective/obico-server/blob/471c0364c7c823194bc9c5c2b13e76aa1ebad6b9/backend/lib/syndicate.py#L29
However I have no idea how to fix it, since I haven't used Django.

kennethjiang, 2024-10-31:
Since we store the full url in the db (there was a long discussion on why we can't store only partial url), we can choose only one site in creating "img_url". I do agree this is a problem for users running multiple sites.

Do you see any solutions?

whc2001, 2024-10-31:
> Since we store the full url in the db (there was a long discussion on why we can't store only partial url), we can choose only one site in creating "img_url". I do agree this is a problem for users running multiple sites.
> 
> Do you see any solutions?

I'd say maybe a environment variable to forcefully rewrite the primary URL to mitigate?

It does not solve the real problem of multiple sites though, as in my configuration the other two sites (on port 15853 and 15854) are for tunnel usage, I am still only running a single site configuration (port 48877).

kennethjiang, 2024-11-10:
I'm pretty sure we have had discussions on discord server about this issue. Closing it now.

closed 2024-11-10 by kennethjiang