TheSpaghettiDetective/obico-server #1052: [Feature] Docker Compose file that is compatible with Portainer

Boerny41, 2024-12-28:
**Describe the solution you'd like**
Can we get a docker compose file that is compatible with container management software like portainer? The current one gives out an error.
`Failed to deploy a stack: Service ml_api Building Service tasks Building unable to prepare context: path "/data/compose/64/backend" not found`

kennethjiang, 2025-01-02:
Thank you for making this feature request. If anyone can submit a PR I'll be happy to review and merge it.

DeathCamel58, 2025-01-24:
I got this running in Portainer by manually creating the images, then using them in the `docker-compose.yml`.

To build the images:
1. Download and extract this repository
2. Create a tar archive containing the `ml_api` folder
3. Create a tar archive containing the `backend` folder
4. In Portainer, go to `Images`, then click `Build a new image`
5. For name, input `obico-ml_api`
6. Select `Upload`
7. Click `Select file`, and select the archive of the `ml_api` folder
8. Click `Build the image`
9. Wait for it to complete
10. Go back to `Images` -> `Build a new image`
11. For name, input `obico-backend`
12. Select `Upload`
13. Click `Select file`, and select the archive of the `backend` folder.
14. Click `Build the image`
15. Wait for it to complete

To use the images in a stack:
1. In this repository's docker compose, replace
    ```
    build:
      context: backend
      dockerfile: 'Dockerfile'
    volumes:
      - ./backend:/app
      - ./frontend:/frontend
    ```
    With:
    ```
    image: obico-backend:latest
    volumes:
      - obico-backend:/app
      - obico-frontend:/frontend
    ```
2. Replace:
    ```
    build:
      context: ml_api
    ````
    With:
    ```
    image: obico-ml_api:latest
    ```
3. Add this to the end of the compose file:
    ```
    volumes:
      obico-backend:
      obico-frontend:
    ```

At this point, the stack will start, but you'll get an HTTP 500 error per #1018. Opening a shell on `obico-dev-web-1` and running `/manage.py site --add server_ip_address:server_port` (replacing your server IP address and port) will get you to the django web UI.

jcolladosp, 2025-01-27:
> I got this running in Portainer by manually creating the images, then using them in the `docker-compose.yml`.
>
> To build the images:
>
> 1. Download and extract this repository
> 2. Create a tar archive [...]

How do you manage to update the images when there's a new release?

DeathCamel58, 2025-01-27:
I believe you can re-build an image with the same name, then open the stack, go to the editor tab for the stack, and there should be a button called `Update the stack`.

jcolladosp, 2025-01-27:
> I believe you can re-build an image with the same name, then open the stack, go to the editor tab for the stack, and there should be a button called `Update the stack`.

Yes, but you also need to rebuild the .tar files, isn't it?

DeathCamel58, 2025-01-27:
Yes. You would need to rebuild the tar files. I'd imagine that it's possible to build the images using the `URL` option instead of the `Upload` option to have Portainer clone the repository, but I haven't tried that. I wouldn't like the idea of cloning the whole repository for each of the two images.

DeathCamel58, 2025-01-27:
Filed #1055 to try to get them to upload official prebuilt container images to a registry. This will allow us to just use their latest in a stack, so that we can easily get updates.

kennethjiang, 2025-01-27:
I will use this issue as the main issue since there are already discussions here.

A community member previously took a stab at it. He broke the work into 2 pieces: https://github.com/TheSpaghettiDetective/obico-server/pull/824 and https://github.com/TheSpaghettiDetective/obico-server/pull/821. I (prematurely)  merged #824 believing that #821 will be ready soon. But eventually had to revert #824 because #821 was eventually abandoned.

The Obico cloud uses Kubernetes to run a scalable clusters. Hence we expect the self-hosted community to implement this feature because we ourselves won't be the users of this feature. I hope this makes sense.

d-mcknight, 2025-07-03:
> The Obico cloud uses Kubernetes to run a scalable clusters. Hence we expect the self-hosted community to implement this feature because we ourselves won't be the users of this feature. I hope this makes sense.

Would you be open to a PR to implement an image build pipeline in GHA similar to #824 ? If yes, should PR's be based off the `master` branch? I didn't see these details in the [contributing guide](https://www.obico.io/docs/developer-guides/contribute/#our-source-code-repos)

kennethjiang, 2025-07-03:
> Would you be open to a PR to implement an image build pipeline in GHA similar to [#824](https://github.com/TheSpaghettiDetective/obico-server/pull/824) ? If yes, should PR's be based off the [...]

Yes please send PR. Just make sure you test it before you set the PR to be ready for review.  Also I just updated the contributing guidelines. It may still not be clear enough. Feel free to ask if it's the case.

d-mcknight, 2025-07-03:
Looks good to me; I got a decent start on this last night and will open a draft PR later today once I confirm at least minimum functionality is there

md1986, 2025-08-03:
> I got this running in Portainer by manually creating the images, then using them in the `docker-compose.yml`.

Followed your guide and got the web server up and running. Unfortunately when trying to add the Django site it seems that the frontend part is missing. Could you please check if something changed since you wrote it?