TheSpaghettiDetective/obico-server #1056: [BUG] "localhost:3334" instead of real webcam address in web frontend.

DBa20-24, 2025-01-28:
**Describe the bug**
After adding a printer (QIDI Plus4), the app shows as it should, but the web interface only shows a symbolic picture instead of webcam image.

Inspecting the web page, I found that the "<img>" element showing the picture had the src attribute of "https://localhost:3334/media/tsd-pics/snapshots/2/latest_unrotated.jpg?digest=...."

Replacing the "localhost:3334" by the real address (in my case, a reverse proxy address) fixed the issue in terms of I was able to see the screenshot. Obviously, I cannot replace anything in the webpage.

**To Reproduce**
Steps to reproduce the behavior:
1. Blank-install obico-server
2. Install obico with KIAUH on the printer and add it
3. Access the web interface of obico-server and have a look at the camera picture.

**Screenshots**

[image]

[image]

jaketheduque, 2025-01-30:
Also having this issue. I've tried changing the INTERNAL_MEDIA_HOST variable in docker-compose.yml to my domain name to see if that fixes it, but seems like that isn't the problem.

jaketheduque, 2025-01-30:
Got it working. The URL the frontend uses depends on configured 'sites' in the Django admin page.

What did it for me was:
1) Removing all sites **except** for the one that I wanted the front end to use to access media
2) Restart using `docker compose down && docker compose up -d` (`docker compose restart` may work as well, I just did the two separate steps as a precaution)
3) Double check that the media is loading on the webpage

**Note:** I tried adding back the other domains but the URL on the website changes back to the wrong one. It might be whatever the first one listed under the 'sites' tab is. Is this intended functionality? I'm not familiar with Django so I'm not sure how it is supposed to work.

kennethjiang, 2025-02-02:
There are plenty of discussions around this issue on our discord server https://obico.io/discord/. Please go there to contribute to the discussion.

I'm closing this issue as our github issues are only for confirmed bug and feature requests.

closed 2025-02-02 by kennethjiang