TheSpaghettiDetective/obico-server #1063: [BUG] No failure detection even at highest settings

charredchar, 2025-03-23:
> **Note**
> Most of the times, an issue you may potentially run into is not a bug. Instead, it is related to configurations, such as firewall rules, incompatible docker version, reverse proxy setups, etc.
> 
> If you are not sure, please don't open a bug report here. Instead, you can get the issue solved much faster by seeking help in our support forum: https://obico.io/discord.
> 
> Proceed to opening a bug report only if you are sure it is a defect in the code given a correct configuration.

I attempted to join the Discord server but I get an "Unable to accept invite" error.
I am not sure if it is a configuration issue or not but I had it working in the past but now it is not working.

**Describe the bug**
Running a print I knew would have a failure to see if Obico would properly alert me and to tune the sensitivity. But it didn't show any signs of finding a failure.

**Screenshots**

[image]

**Hosting environment (please complete the following information):**
 - OS: Ubuntu 20.04.6 LTS on a Hyper-V VM
 - Version: I can't actually find where this is listed?
 - Hardware: 2x Intel Xeon E5-2650 v2, 192GBs of RAM, 40GB SSD

**Additional context**
I had the server functioning successfully some time last year before I had to shut it down for a move. After the move I had also redone my printer config after some changes to the printer and needed to re-add it. Everything appears to be functional, the WebUI says it is watching but it never marked this print as a failure or even any signs of it being suspicious like I've seen it do in the past.

kennethjiang, 2025-03-24:
I just tested the link https://obico.io/discord with a new discord account and it worked for me. Can you try it again or post a screenshot for the discord error here? Discord is quite important as I don't respond to github issues that quickly.

For now, you can try `docker compose logs -f` to see if there are errors to debug more.

kennethjiang, 2025-03-26:
There are a lot of errors in your log file. it seems that the db is somehow corrupt. maybe you should erase everything and start over.

charredchar, 2025-03-27:
I was afraid something like that might be the case... It'll take me a little time to reinstall, I'll probably just do a new VM to be safe (as you can tell by the name the current one is pretty old) and report back if I have any of the same issues.

Is there a way to test failure detection that doesn't involve a real print? I'm not a huge fan of trying to force a print to fail. lol And contrary to the first post I rarely have failures which is why I went a little bit before noticing it wasn't catching them. heh

kennethjiang, 2025-03-31:
> I was afraid something like that might be the case... It'll take me a little time to reinstall, I'll probably just do a new VM to be safe (as you can tell by the name the current one is pretty old) [...]

The only way that I'm aware of is to use the virtual printer in octoprint to simulate a print.

kennethjiang, 2025-05-23:
closing as it seems like an one-off env issue.

closed 2025-05-23 by kennethjiang

charredchar, 2025-05-31:
Sorry I let this go for so long as I was attempting to make a virtual printer and find some way of sending what would look like a failed print to Obico... I haven't quite got it working yet.

I was able to get it back up and running after a reinstall and even confirmed it by forcing a failure so yes, it was likely a one-off env issue. Unfortunately after a while on the interface the needle once again doesn't move even at the most sensitive setting. I check the logs and find no errors then force a failure to find the needle does move but I have pretty much lost confidence in Obico without some way to periodically test it.

Would you happen to have any information you can link for "simulating a failed print" for Obico? It would be really nice if there was some way to "ping" the server (like by injecting a known failure snapshot for the model to analyze and respond to) before a print starts. Even if I needed to send something via command line pointing to an image either on the printer host or on the server, scheduling it to do that like once a week or something would be a piece of mind knowing there is no issue compared to dealing with a virtual printer needing to check manually. What's stopped me at the virtual printer part so far is getting images for Obico to analyze. I was successful in setting up the virtual printer (with OctoPrint in a VM) and linking it but a simple test image for the "web camera" isn't working. I could attempt streaming a video in place of an image but I also can't find any non-time-lapse video examples online of failures for tests.

kennethjiang, 2025-05-31:
> Sorry I let this go for so long as I was attempting to make a virtual printer and find some way of sending what would look like a failed print to Obico... I haven't quite got it working yet.
>
> I was [...]

The most reliable way to test is to run a small print that’s guaranteed to fail—such as one with severe overhangs and no supports.

charredchar, 2025-06-13:
That's disappointing as it also means I need to tie up my printer and can't run any kind of test *before* a print starts. I saw a couple of random issues about some kind of testing but it seems none of that had gone anywhere or is viable.
Sadly, as I said before, I don't think I can trust this software anymore without some kind of verification testing in place. I get that might not be useful for people relying on the cloud service but for us wanting to self-host this is a deal breaking.

kennethjiang, 2025-06-13:
> That's disappointing as it also means I need to tie up my printer and can't run any kind of test _before_ a print starts. I saw a couple of random issues about some kind of testing but it seems none [...]

I haven't heard such point from any other self-hosted users. But if this is your concern, feel free to fork the repo and make the changers you want. I made the source code available so that anyone who has a special need can customize the software as they want. I don't owe anyone to change the software to cater to their special need, and I hope this is a reasonable position.