TheSpaghettiDetective/obico-server #1064: [BUG] failed to build Container on New Server

fmontebaur, 2025-03-29:
Docker Compose canceled:

ERROR [tasks 3/5] RUN pip install -U pip pipenv==2022.12.19

CANCELED [ml_api 4/9] RUN pip install --upgrade pip

 > [tasks 3/5] RUN pip install -U pip pipenv==2022.12.19:
1.500 Requirement already satisfied: pip in /usr/local/lib/python3.10/site-packages (24.3.1)
6.507 WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f2b34288d60>: Failed to establish a new connection: [Errno -3] Try again')': /simple/pip/

76.59 ERROR: Could not find a version that satisfies the requirement pipenv==2022.12.19 (from versions: none)
81.63 ERROR: No matching distribution found for pipenv==2022.12.19

**Hosting environment (please complete the following information):**
 - OS: Debian 12 / Docker
 - Version: Community 28.0.4
 - Hardware Proxmox Virtual Maschine

kennethjiang, 2025-03-31:
This looks like an environment issue instead of a bug. Please come to our discord server https://obico.io/discord/ to seek help.

kennethjiang, 2025-03-31:
Closing it for now. feel free to reopen it if you have a reason to believe this is a bug.

closed 2025-03-31 by kennethjiang