TheSpaghettiDetective/obico-server #1077: [BUG] password reset E-Mail not working

leder11011, 2025-06-09:
**Note**
Most of the times, an issue you may potentially run into is not a bug. Instead, it is related to configurations, such as firewall rules, incompatible docker version, reverse proxy setups, etc.

If you are not sure, please don't open a bug report here. Instead, you can get the issue solved much faster by seeking help in our support forum: https://obico.io/discord.

Proceed to opening a bug report only if you are sure it is a defect in the code given a correct configuration.

**Describe the bug**
Password reset E-Mail doesn't open new password page in app obico 

**To Reproduce**
Steps to reproduce the behavior:
1. Click password reset
2. Enter valid email address 
3. Click link in received email 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hosting environment (please complete the following information):**
 - OS: android 
 - Version 15
 - Hardware nothing phone 2

**Additional context**
Add any other context about the problem here.

kennethjiang, 2025-06-09:
https://github.com/user-attachments/assets/eb1e53ff-eff1-449e-b58f-30d25930d0a8 

Cannot reproduce. Screen recording is attached.

leder11011, 2025-06-09:
when talk is in vain you have to learn through experience