TheSpaghettiDetective/obico-server #1090: [BUG] Docker containers obico-server-web and obico-server-tasks fail with ModuleNotFoundError: No module named 'langfuse'

spitfire, 2025-07-17:
**Note**
I used to have a working Obico server, after shutting down and doing git pull on the 'release' branch it does not work anymore.

My current docker-compose override file:

```
cat docker-compose.override.yml
services:
  web:
    # build:
    #   context: web
    environment:
      OCTOPRINT_TUNNEL_PORT_RANGE: "15853-15858"
    ports:
      - "15853:3334"
      - "15854:3334"
      - "15855:3334"
      - "15856:3334"
      - "15857:3334"
      - "15858:3334"
    volumes:
      - obico:/app/backend/static_build/media
    labels:
      - "com.centurylinklabs.watchtower.enable=false"

  ml_api:
    # enables GPU access for container
    # build:
    #   context: ml_api
    deploy:
      resources:
        reservations:
          devices:
            - driver: nvidia
              count: 1
              capabilities: [gpu]
    volumes:
      - obico:/app/backend/static_build/media
    labels:
      - "com.centurylinklabs.watchtower.enable=false"

  tasks:
    # build:
    #   context: tasks
    volumes:
      - obico:/app/backend/static_build/media
    labels:
      - "com.centurylinklabs.watchtower.enable=false"

volumes:
  obico:
    driver_opts:
      type: "nfs"
(nfs server address an path removed)
```

**Describe the bug**
```
docker logs -f obico-server-tasks-1
celery.utils.dispatch.signal ERROR    Signal handler <bound method DjangoFixup.on_import_modules of <celery.fixups.django.DjangoFixup object at 0x730cd32099f0>> raised: ModuleNotFoundError("No module named 'langfuse'")
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/celery/utils/dispatch/signal.py", line 276, in send
    response = receiver(signal=self, sender=sender, **named)
  File "/usr/local/lib/python3.10/site-packages/celery/fixups/django.py", line 97, in on_import_modules
    self.worker_fixup.validate_models()
  File "/usr/local/lib/python3.10/site-packages/celery/fixups/django.py", line 137, in validate_models
    run_checks()
  File "/usr/local/lib/python3.10/site-packages/django/core/checks/registry.py", line 88, in run_checks
    new_errors = check(app_configs=app_configs, databases=databases)
  File "/usr/local/lib/python3.10/site-packages/django/core/checks/urls.py", line 42, in check_url_namespaces_unique
[... 124 lines cut ...]
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/app/app/jusprin/urls.py", line 5, in <module>
    from . import views as jusprin_views
  File "/app/app/jusprin/views.py", line 9, in <module>
    from langfuse.decorators import langfuse_context, observe
ModuleNotFoundError: No module named 'langfuse'
```

**To Reproduce**
Steps to reproduce the behavior:
1. docker compose down
2. git pull
3. docker compose up -d

**Hosting environment (please complete the following information):**
 - OS: Debian Bookworm
 - Version: 12.11
 - Hardware: PC, Nvidia RTX 2070

**Additional context**
Used to work before update

kennethjiang, 2025-07-18:
This is because you didn't use the correct command to update the server: https://www.obico.io/docs/server-guides/ops/#upgrade-obico-server

closed 2025-07-18 by kennethjiang

spitfire, 2025-07-18:
Do you mean `git checkout release`? I did that as well, didn’t help. pre-realised branch didn’t work for me either.

spitfire, 2025-07-19:
> docker compose up --build -d

Thanks, I was missing the `--build` flag, was it in the docs before?
Would it be possible for you to provide pre-built/tagged docker images?

kennethjiang, 2025-07-21:
We have a PR to create a pre-built image: https://github.com/TheSpaghettiDetective/obico-server/pull/1086